merbanan/rtl_433 #3307: Bugs in tpms_abarth124.c and tpms_renault_0435r.c

KlausPetersEnreach, 2025-06-22:
Hello.

Both devices "Abarth 124 Spider TPMS" and "Renault 0435R TPMS" are using FSK 9 byte Manchester encoded TPMS with xor checksum.
Both devices accept a preamble of size "55 55 56".

But afterwards the simple check is

tpms_abarth124.c
```
    // check checksum (checksum8 xor)
    checksum = xor_bytes(b, 9);
    if (checksum != 0) {
        return 0; // DECODE_FAIL_MIC;
    }
```
tpms_renault_0435r.c 
```
    // check checksum (checksum8 xor)
    int chk = xor_bytes(b, 9);
    if (chk != 0) {
        return DECODE_FAIL_MIC;
    }
```
These checks ignore the position of the checksum byte!

This should be corrected to something like this:

tpms_abarth124.c
```
    checksum = xor_bytes(b, 8);
    if (checksum != b[8]) {
        return 0; // DECODE_FAIL_MIC;
    }
```
tpms_renault_0435r.c 
```
    int chk = xor_bytes(b, 7) ^ b[8];
    if (chk != b[7]) {
        return DECODE_FAIL_MIC;
    }
```
Thanks in advance!

KlausPetersEnreach, 2025-06-22:
Please try it out
> .\rtl_433-rtlsdr.exe -y "{178}55555556999aa5a6a9aaa55599a5665955559a6a5555"

zuckschwerdt, 2025-06-22:
It would be a good idea for any other checksum method, but as pointed out if `chk = xor_bytes(b, N)` then for any imagined `shuffle_bytes(b, N)` it still holds that `chk == xor_bytes(b, N)` -- the order of bytes does not matter.
(`chk` is likely `0` but could be any other fixed "init" value.)

KlausPetersEnreach, 2025-06-23:
Yes, you both are completely right.
After intensive testing I figured out that it makes no difference.
This is the nature of a simple xor.

Sorry for my complains and thank you for your patience.

I'll close this issue, because it was nonsense.

closed 2025-06-23 by KlausPetersEnreach

zuckschwerdt, 2025-06-23:
A chance to improve or clarify things is always worth taking. Thank you for your interest in the project!