merbanan/rtl_433 #3312: WMBus Reception Somehow Incompatible With Other Devices

luarvique, 2025-06-28:
When using rtl_433 within OpenWebRX, it has been found that feeding rtl_433 with the 250kHz IQ stream receives most ISM devices, but fails receiving Wireless M-Bus (EN 13757-4) signals. When changing the incoming IQ stream to 1.2MHz, WMBus reception starts working, but rtl_433 stops receiving other ISM devices.

The most interesting part is that in **both** cases we pass the input signal through a 250kHz bandpass filter. So, in principle, nothing should change for the rtl_433.

Can something be changed in the rtl_433 invocation to make it receive all the devices at the same sample rate? Or maybe it is a bug inside rtl_433?

luarvique, 2025-06-28:
rtl_433 is being invoked as follows:

`rtl_433 -r cs16:- -s <sampleRate> -M time:unix -F json -A`

luarvique, 2025-06-28:
> I think the reason is that >868MHz the fsk minmax demod is used by default. In theory we could run both demods but some one would have to implement this.

Could you elaborate [to someone unfamiliar with rtl_433 internals] on what exactly goes sideways in the above scenario?

zuckschwerdt, 2025-06-29:
Regarding your 250kHz bandpass filter: increasing the sample rate also increases the time resolution.
The time resolution needs to somewhat match the signal speed. I.e. you can't demod the 100 kb/s (10 µs) data with a 250 sample rate (resolution 4 µs, not enough oversampling). The other way around too high of a sample rate will hinder the demod to track data bits -- seeing a level for too long (akin a DC component) will "null out" the level discriminator.

luarvique, 2025-06-29:
In your opinion, what would be the optimal sampling rate allowing rtl_433 to receive the most signal sources?

luarvique, 2025-07-01:
So, I have done some experimentation to determine what makes or breaks rtl_433 reception for me. Unfortunately, we do not have WMBus here, so all experiments were done at 433MHz. Also, switched rtl_433 to CF32 input to avoid conversion:

`rtl_433 -r cf32:- -s 250000 -M time:unix -F json -A`

**Results:**
— AGC seems to be necessary, it won't detect signals unless input IQ stream passes through AGC.
— Having 250kHz bandpass improves reception (when compared with just using 250ksps IQ stream).
 — When using 1.2Msps IQ, only strongest (like, absolutely strongest) signals are detected at 433MHz.
— Switching from the "classic" decoder to "minmax" also drops the success rate at the 433MHz to just one signal source (with everything else being equal, i.e. 250ksps, 250kHz bandpass).

I will try coming up with some sample data for you, although given the required size, it might be difficult.

luarvique, 2025-07-01:
Samples (2 x 130MB gzipped)

[250ksps CF32 stream that works](https://www.dropbox.com/scl/fi/hgezj6ifwn4y135ovkg9p/rtl-433-250ksps-cf32.bin.gz?rlkey=2ha74ur8ggqqa12c1eqocznby&st=g5zf2qd6&dl=0)

[1200ksps CF32 stream that does not work](https://www.dropbox.com/scl/fi/n2f53sqnzw8xe62riuvol/rtl-433-1200ksps-cf32.bin.gz?rlkey=x5m0x0mks6zz7fv7ozm4u4luh&st=epqf6wwy&dl=0)

luarvique, 2025-07-01:
If it matters, they are from the same exact environment and hardware setup, with the same three main ISM sources (I think), same 250kHz bandpass applied before feeding IQ to the rtl_433. The only difference is the sample rate. The higher rate sample is obviously shorter, but should still include pings from at least two ISM sources.

zuckschwerdt, 2025-07-01:
Yes, the recordings look the same, I can spot the transmissions. It's a good example and puzzling why reception is so much worse.

Is that AGC a digital signal boost? It might slope to fast or quantize too much and then not play well with our noise tracking.

luarvique, 2025-07-01:
> Is that AGC a digital signal boost? It might slope to fast or quantize too much and then not play well with our noise tracking.

Here is the AGC:
https://github.com/luarvique/csdr/blob/master/src/lib/agc.cpp

Using these defaults:
```
            float reference = 0.8;
            float attack_rate = 0.1;
            float decay_rate = 0.001;
            float max_gain = 65535;
            unsigned long int hang_time = 200;
            float gain_filter_alpha = 1.5;
```

luarvique, 2025-07-01:
Here are samples with AGC module removed and "-Y autolevel" enabled. Notice how well GZIP compresses them now: where previously it could not achieve higher than 50% compression, it now compresses them to mere hundreds of kilobytes. The end result is still the same though: the 1.2Msps data only led to ONE decode, while the 250ksps data had many.

[rtl-433-1200ksps-cf32-noagc.bin.gz](https://github.com/user-attachments/files/21008747/rtl-433-1200ksps-cf32-noagc.bin.gz)
[rtl-433-250ksps-cf32-noagc.bin.gz](https://github.com/user-attachments/files/21008746/rtl-433-250ksps-cf32-noagc.bin.gz)

zuckschwerdt, 2025-07-02:
Both files look nearly empty. What hardware did you record this on? (and wouldn't the native format be cs16?) Is the noise really at -inf dB?

There is a diagnostic to show I/Q, AM and logic states: `rtl_433 -w filename.sr filename.cf32` then open with SigRok's Pulseview. But there is no clear way to use that to improve things.

luarvique, 2025-07-02:
OpenWebRX uses complex floats internally, hence the CF32 format. I am assuming that what made them "empty" (and did wonders to compression ratio) is the lack of AGC before they arrive into rtl_433. Still, "rtl_433 -Y autolevel" decodes stuff from the 250ksps stream, while failing on the 1.2Msps one. So, there must be some difference.

zuckschwerdt, 2025-07-03:
There is now a quickly hacked version at https://triq.org/pdv4/ (careful with file sizes, there is a 5x buffer overhead).
The "Amplitude bar" option can be used to show the ASK discriminator internals:
- black is the amplitude
- blue is the filtered envelope
- green is the low estimate
- red is the high estimate
- background color shows decided pulses

E.g. here is a complicated situation (should be 7 equal width pulses, maybe)

[image]

zuckschwerdt, 2025-07-04:
The "CU8" in the pdv infos wasn't fully implemented, it's fixed now. I now need to figure out how to implement "autolevel" for the pdv display, otherwise the default discriminator levels (-12 dB) are way out of range.

luarvique, 2025-07-04:
Here is the current comparison. Two things are strange:
1) The 1.2Msps sample is still being treated as 250ksps one.
2) There are no pulses being decoded underneath both spectrograms (shouldn't there be some at least for the 250ksps sample?)

[image]

zuckschwerdt, 2025-07-04:
Metadata is picked up from the filename but seems it isn't fully implemented. rtl_433 will accept:
"M", "MHz", "Hz" and "k", "ksps", "sps" but pdv is missing the hz and sps ones. I'll add that now.

The analyzer in pdv is lacking the auto-level feature and needs a fixed minimum of -12 dB to trigger the decoding. I'm working on adding an auto-level feature.

zuckschwerdt, 2025-07-04:
Metadata in pdv can now parse with "Hz" and "sps" suffix.

zuckschwerdt, 2025-07-07:
I've done some experimenting with our discriminator. It seems the issue is that without noise and at such low levels the estimates don't track low enough. Higher sample rates might worsen this as changes are less steep then.

How are these recordings preprocessed? It seems there is near zero noise and every amplitude is part of a signal. That would be a great if we could achieve it reliably. Or is it that the gain is (by chance?) tuned to cut off just above the noise floor when converting f32 to s16? I.e. are the (at most 12 bit ADC) values aligned to some small float value (instead of 0 dB at 1.0)?

To me it seems that the signal without the AGC you applied is preferable for further processing.

luarvique, 2025-07-07:
The full data path is as follows:

1) The original IQ stream is received with an RSPdx with 12bit resolution, at 6Msps.

2) The 250ksps CF32 IQ stream at a target frequency (433.92MHz) is cut out of the original stream using decimation.

3) The decimated CF32 IQ stream is filtered to [-125kHz..+125kHz] using a FIR filter.

4) The CF32 IQ stream is then fed directly to rtl_433, as follows:
`rtl_433 -r cf32:- -s 250000 -M time:unix -F -A -Y autolevel -W cf32:/tmp/output.cf32`

5) The output.cf32 files for 250ksps and 1.2Msps sample rates are attached above.

I am guessing that the perceived lack of noise might be due to decimation applied as described above.

zuckschwerdt, 2025-07-07:
Thanks for the details! Interesting, I'll need to analyze the levels more in depth, sounds like the whole ~12~14 bit should be still there? The decimation (dropping samples) will not affect levels.

The RSPdx is native CS16, with 14bit resolution [MSB-aligned](https://github.com/pothosware/SoapySDRPlay3/blob/master/Streaming.cpp#L42), and has CF32 [aligned to 1.0](https://github.com/pothosware/SoapySDRPlay3/blob/master/Streaming.cpp#L151C36-L151C41). Is the filter maybe not normalized to 1.0?

Is a 250 kHz (-125kHz..+125kHz) low-pass filter effective at 250 kHz sample rate? The roll-off would be starting from the edges of the 250 kHz and thus be outside the available bandwidth, right? Measured in the 1200 ksps recording the signal is ~256 kHz wide. A very good filter but I think not practical for 250 kHz? It's working for the 1200 ksps signal though, which I gather is what you prefer to send to rtl_433.

luarvique, 2025-07-07:
Sorry for the 12/14bit confusion. Yes, the data is received in the RSP's native resolution. I am currently not aware how exactly filtering is done in OpenWebRX: this would require digging deep into the pipeline.

As to the 250kHz filter, I have verified that it **does somehow improve** the rtl_433 reception, even at the 250ksps rate. I will try providing you with 250ksps rrecordings with vs without the filter.

zuckschwerdt, 2025-07-07:
In your examples the noise level seems to sit at -60 dB and fades out below -75 dB which is still in the original range for 14 bit (~85 dB). Your files seem to show the RSPdx signal with unchanged levels. And rtl_433 should be able to work with those levels, the processing is at 16 bit.
Disregard my concerns about the levels, it should be okay without boosting the signal.

gdt, 2025-07-16:
My take on this issue is:
1) belongs as Discussion not issue
2) there are likely some feature tickets that could be filed based on the insights so far.

I'll convert to discussion >= 24h absent rationale that it belongs as an issue.

gdt, 2025-07-16:
My preference, if it's ok with you, is for you to open a new ticket that explains the discriminator issue, point to this, and we can keep the other one open and close this.  I'm fine with feature tickets being open, but I would like them to be crisp without a ton of discussion so that one can look at them and say "oh, that's a feature, this is what it's about, and I don't need to spend 10 minutes to figure it out".   Basically i see this discussion as having been useful to surface a place where the code needs improving, but mostly people don't need to read how we got here.

closed 2025-07-17 by zuckschwerdt