merbanan/rtl_433 #3313: Doorbell Smartcon Kinetik WD-150

whitevast, 2025-06-29:
Good afternoon! I'm trying to create a Smartcon Kinetik WD-150 doorbell interceptor. I'm receiving clear, repetitive signals, but I can't figure out how to intercept the event because there are three different pulse lengths. I'd appreciate any assistance.
Here are the signals from several devices:
https://triq.org/pdv/#AAB01107010050013C00B4003003F4003C27148455+AAB01E07010050013C00B4003003F4003C27148590A0B090B0B0B09090A0B0B0A455+AAB01E07070050013C00B4003003F4003C2714B090A0B090B0B0B09090A0B0B0A455+AAB01107010050013C00B4003003F4003C2714B655
https://triq.org/pdv/#AAB00F06010050013C00B4003003F427148455+AAB01C06010050013C00B4003003F427148390A0B090B0B0B09090A0B0B0A455+AAB01C06040050013C00B4003003F42714B090A0B090B0B0B09090A0B0B0A455+AAB01A06010050013C00B4003003F42714B090A0B090B0B0B09090A0B555
https://triq.org/pdv/#AAB00F06010054013C00B8003003F027148455+AAB01C06010054013C00B8003003F027148390A0B090B0B0B09090A0B0B0A455+AAB01C06040054013C00B8003003F02714B090A0B090B0B0B09090A0B0B0A455+AAB01306010054013C00B8003003F02714B090A0B08555
https://triq.org/pdv/#AAB00F06010054013800B4002C03F027148455+AAB01C06010054013800B4002C03F027148390A0B090B0B0B09090A0B0B0A455+AAB01C06060054013800B4002C03F02714B090A0B090B0B0B09090A0B0B0A455+AAB01A06010054013800B4002C03F02714B090A0B090B0B0B09090A0B555

zuckschwerdt, 2025-06-29:
Interesting signal.
Yes, there are at least three different symbols. Could be decoded with PCM, but that's way to much data for just the ID of a doorbell.
It's not what we expect from a doorbell. We'd expect to see two different symbols (pulse lengths or gap lengths) and usually 24 bit, see https://triq.org/rtl_433/PULSE_FORMATS.html#pwm-%E2%80%94-pulse-width-modulation

Can you capture and upload a good sample as zip here, that way we can check if the demodulation is correct. https://triq.org/rtl_433/ANALYZE.html

whitevast, 2025-06-29:
Yes, of course.

[1.ZIP](https://github.com/user-attachments/files/20967698/1.ZIP)

zuckschwerdt, 2025-06-29:
If we look at the samples in https://triq.org/pdv3/ and change the FFT size to e.g. 16 we can make out that there are only two symbols, long and short. The demod might have trouble with two things here: The center frequency is too close to the signal frequency, use e.g. `-f 433.87M` and the signal is too strong try a lower gain or more distance. Or you could maybe increase the sample rate (`-s 1024k`) to get a finer resolution.

ProfBoc75, 2025-06-29:
@whitevast as proposed by @zuckschwerdt you can decode with PCM and apply symbol coding like this:

    rtl_433 -X "n=kinetic_door_bell,m=OOK_PCM,s=48,l=48,r=1100,gap=150,repeats>=3,unique,symbol_zero={3}8,symbol_one={3}e"

whitevast, 2025-06-29:
Yes! It works great!!! Thanks a lot for your help! Now I can integrate this call into my smart home.
I'll try to understand these parameters so that I can catch similar signals in the future.

whitevast, 2025-06-29:
By the way, increasing the sampling frequency also has an effect, and you can even try to decipher the data:
https://triq.org/pdv/#AAB00D0501002700620402001A27108255+AAB0250505002700620402001A27108193938193818193938181818193938193938193818181938255+AAB0250501002700620402001A27108193938193818193938181818193938193938193818181938455

closed 2025-06-30 by whitevast