merbanan/rtl_433 #3323: Feature Request: Add/Improve decoder for ThermoPro TP60 (TX-2B/2C) on 433.867 MHz

djcarter, 2025-07-09:
[ThermoPro-TP60B_Instruction_20MB.pdf](https://github.com/user-attachments/files/21138444/ThermoPro-TP60B_Instruction_20MB.pdf)

Sensor model: ThermoPro TP60 outdoor transmitter (model TX-2B/2C).
Frequency: 433.867 MHz

I have gqrx recordings that visually confirm transmission at intervals of ~50 sec that is matched by the indicator LED on the outdoor unit and used the latest rtl_433 (built from source) to decode, using -A as well as -R protocols (97, 40, 55, 163, 256, 259, 267, 278) without successful decoding.

I have gqrx recording of a [single transmission 2.7MB](https://drive.google.com/file/d/1PZP4E-8OsgxxZ32cFgQyKrjzi3LHDepU/view?usp=drive_link) and [two consecutive transmissions 10.8MB](https://drive.google.com/file/d/1LoS_Bb50sdzbjsqGkTn4672375dtuWUn/view?usp=drive_link) so that you can see the timing between transmissions if that helps.

See the dump of rtl_433 -A below:

```
darren@linux:~$ /usr/local/bin/rtl_433 -r gqrx_20250709_085041_433867800.wav -f 433.867M -s 2.048M -A
rtl_433 version master-2-g802895dd branch master at 202507072052 inputs file rtl_tcp RTL-SDR with TLS
[Input] Test mode active. Reading samples from file: gqrx_20250709_085041_433867800.wav
Detected FSK package	@0.000500s
Analyzing pulses...
Total count: 1141,  width: 87.21 ms		(178613 S)
Pulse width distribution:
 [ 0] count:  148,  width:   15 us [12;18]	(  30 S)
 [ 1] count:  157,  width:   32 us [26;40]	(  66 S)
 [ 2] count:  143,  width:   21 us [19;26]	(  44 S)
 [ 3] count:  135,  width:    7 us [6;9]	(  15 S)
 [ 4] count:  101,  width:   84 us [68;108]	( 173 S)
 [ 5] count:   39,  width:  183 us [149;234]	( 375 S)
 [ 6] count:  140,  width:   50 us [41;62]	( 102 S)
 [ 7] count:   45,  width:  127 us [107;153]	( 261 S)
 [ 8] count:  108,  width:   10 us [9;12]	(  21 S)
 [ 9] count:   19,  width:   64 us [60;73]	( 132 S)
 [10] count:   95,  width:    5 us [5;6]	(  10 S)
 [11] count:    9,  width:  265 us [239;308]	( 542 S)
 [12] count:    2,  width:  373 us [343;403]	( 763 S)
Gap width distribution:
 [ 0] count:  159,  width:   11 us [9;14]	(  23 S)
 [ 1] count:  168,  width:   24 us [20;30]	(  49 S)
 [ 2] count:  153,  width:    7 us [6;9]	(  15 S)
 [ 3] count:  123,  width:    5 us [5;6]	(  10 S)
 [ 4] count:  148,  width:   16 us [14;20]	(  33 S)
 [ 5] count:  124,  width:   54 us [40;67]	( 110 S)
 [ 6] count:   95,  width:   36 us [30;43]	(  74 S)
 [ 7] count:   76,  width:   98 us [79;125]	( 201 S)
 [ 8] count:   35,  width:   71 us [58;79]	( 145 S)
 [ 9] count:   28,  width:  142 us [117;168]	( 290 S)
 [10] count:   28,  width:  213 us [165;295]	( 437 S)
 [11] count:    2,  width:  333 us [332;336]	( 683 S)
 [12] count:    1,  width:  489 us [489;489]	(1001 S)
Pulse+gap period distribution:
 [ 0] count:  208,  width:   33 us [25;41]	(  68 S)
 [ 1] count:  263,  width:   70 us [56;87]	( 143 S)
 [ 2] count:  173,  width:   48 us [39;56]	(  99 S)
 [ 3] count:   97,  width:   22 us [18;26]	(  45 S)
 [ 4] count:  185,  width:  108 us [87;136]	( 222 S)
 [ 5] count:  106,  width:  167 us [136;212]	( 343 S)
 [ 6] count:   57,  width:   15 us [12;18]	(  30 S)
 [ 7] count:    4,  width:   11 us [10;12]	(  22 S)
 [ 8] count:   40,  width:  246 us [208;301]	( 504 S)
 [ 9] count:    6,  width:  355 us [322;408]	( 728 S)
 [10] count:    1,  width:  494 us [494;494]	(1012 S)
Gap+pulse period distribution:
 [ 0] count:   76,  width:   18 us [14;21]	(  36 S)
 [ 1] count:  251,  width:   47 us [37;60]	(  96 S)
 [ 2] count:  246,  width:   72 us [56;89]	( 147 S)
 [ 3] count:  170,  width:   28 us [22;35]	(  58 S)
 [ 4] count:  164,  width:  106 us [88;133]	( 217 S)
 [ 5] count:  115,  width:  161 us [132;205]	( 330 S)
 [ 6] count:   41,  width:   36 us [32;38]	(  73 S)
 [ 7] count:    3,  width:   83 us [79;87]	( 171 S)
 [ 8] count:   19,  width:   12 us [10;14]	(  25 S)
 [ 9] count:    1,  width:  133 us [133;133]	( 273 S)
 [10] count:   41,  width:  245 us [200;303]	( 501 S)
 [11] count:    7,  width:  348 us [316;408]	( 712 S)
 [12] count:    5,  width:   22 us [22;22]	(  45 S)
 [13] count:    2,  width:  456 us [418;494]	( 934 S)
Timing distribution:
 [ 0] count:  327,  width:   15 us [12;18]	(  30 S)
 [ 1] count:  287,  width:   32 us [26;40]	(  65 S)
 [ 2] count:  285,  width:   21 us [19;26]	(  44 S)
 [ 3] count:  286,  width:    7 us [6;9]	(  15 S)
 [ 4] count:  178,  width:   85 us [68;108]	( 174 S)
 [ 5] count:   68,  width:  183 us [149;234]	( 374 S)
 [ 6] count:  264,  width:   50 us [40;62]	( 102 S)
 [ 7] count:   88,  width:  125 us [106;153]	( 255 S)
 [ 8] count:  209,  width:   10 us [9;12]	(  21 S)
 [ 9] count:   48,  width:   64 us [60;73]	( 132 S)
 [10] count:  218,  width:    5 us [5;6]	(  10 S)
 [11] count:   18,  width:  258 us [230;308]	( 528 S)
 [12] count:    4,  width:  353 us [332;403]	( 723 S)
Level estimates [high, low]:  22157,  12557
RSSI: 1.3 dB SNR: 1.2 dB Noise: -1.2 dB
Frequency offsets [F1, F2]:    7833,  -5143	(+244.8 kHz, -160.7 kHz)
Guessing modulation: No clue...

Detected FSK package	@0.133952s
Analyzing pulses...
Total count:   98,  width: 22.29 ms		(45658 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   23,  width:    9 us [7;11]	(  18 S)
 [ 2] count:    5,  width:   22 us [21;23]	(  46 S)
 [ 3] count:    6,  width:   42 us [37;49]	(  85 S)
 [ 4] count:   29,  width:    5 us [5;7]	(  11 S)
 [ 5] count:    4,  width:  134 us [117;142]	( 274 S)
 [ 6] count:   20,  width:   13 us [11;16]	(  27 S)
 [ 7] count:    3,  width:   17 us [17;17]	(  34 S)
 [ 8] count:    2,  width:   29 us [29;29]	(  60 S)
 [ 9] count:    3,  width:   88 us [80;99]	( 180 S)
 [10] count:    1,  width:   58 us [58;58]	( 118 S)
 [11] count:    1,  width: 1048 us [1048;1048]	(2147 S)
Gap width distribution:
 [ 0] count:   10,  width:   15 us [13;19]	(  31 S)
 [ 1] count:   11,  width:    5 us [5;6]	(  11 S)
 [ 2] count:    6,  width:   87 us [72;101]	( 179 S)
 [ 3] count:   17,  width:  226 us [191;280]	( 462 S)
 [ 4] count:    2,  width: 1254 us [1171;1338]	(2569 S)
 [ 5] count:    5,  width:    8 us [7;9]	(  16 S)
 [ 6] count:    5,  width:  165 us [149;174]	( 338 S)
 [ 7] count:    5,  width:   27 us [24;32]	(  56 S)
 [ 8] count:    3,  width:  871 us [819;918]	(1784 S)
 [ 9] count:    8,  width:  512 us [453;597]	(1049 S)
 [10] count:   10,  width:  345 us [292;414]	( 707 S)
 [11] count:    5,  width:   38 us [34;45]	(  78 S)
 [12] count:    5,  width:  122 us [116;131]	( 250 S)
 [13] count:    3,  width:   65 us [64;67]	( 134 S)
Pulse+gap period distribution:
 [ 0] count:    4,  width:   21 us [19;25]	(  43 S)
 [ 1] count:    8,  width:   30 us [26;33]	(  61 S)
 [ 2] count:   14,  width:  134 us [109;163]	( 274 S)
 [ 3] count:   18,  width:  233 us [199;289]	( 478 S)
 [ 4] count:    3,  width: 1191 us [1055;1343]	(2439 S)
 [ 5] count:    8,  width:   91 us [76;104]	( 187 S)
 [ 6] count:   11,  width:   45 us [37;55]	(  92 S)
 [ 7] count:    3,  width:  878 us [825;924]	(1799 S)
 [ 8] count:    8,  width:  521 us [458;611]	(1066 S)
 [ 9] count:   10,  width:  354 us [298;419]	( 724 S)
 [10] count:    3,  width:  178 us [175;183]	( 365 S)
 [11] count:    2,  width:   66 us [63;70]	( 136 S)
 [12] count:    4,  width:   15 us [14;17]	(  31 S)
 [13] count:    1,  width:   10 us [10;10]	(  21 S)
Gap+pulse period distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:    9,  width:   26 us [22;32]	(  54 S)
 [ 2] count:   11,  width:   42 us [38;47]	(  85 S)
 [ 3] count:    6,  width:   56 us [52;65]	( 115 S)
 [ 4] count:    5,  width:   83 us [76;88]	( 169 S)
 [ 5] count:   16,  width:  232 us [199;287]	( 476 S)
 [ 6] count:    3,  width: 1193 us [1054;1346]	(2444 S)
 [ 7] count:   15,  width:  128 us [105;158]	( 262 S)
 [ 8] count:    4,  width:   17 us [16;19]	(  35 S)
 [ 9] count:    5,  width:  179 us [172;184]	( 367 S)
 [10] count:    3,  width:  879 us [826;928]	(1801 S)
 [11] count:    8,  width:  518 us [459;603]	(1060 S)
 [12] count:   11,  width:  351 us [296;419]	( 718 S)
 [13] count:    1,  width:   14 us [14;14]	(  28 S)
Timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   30,  width:    9 us [7;11]	(  18 S)
 [ 2] count:   10,  width:   22 us [19;25]	(  45 S)
 [ 3] count:   13,  width:   39 us [30;49]	(  79 S)
 [ 4] count:   40,  width:    5 us [5;7]	(  11 S)
 [ 5] count:   12,  width:  135 us [116;167]	( 277 S)
 [ 6] count:   31,  width:   14 us [11;17]	(  29 S)
 [ 7] count:    2,  width:   29 us [29;29]	(  60 S)
 [ 8] count:    9,  width:   87 us [72;101]	( 179 S)
 [ 9] count:    4,  width:   63 us [58;67]	( 130 S)
 [10] count:    3,  width: 1186 us [1048;1338]	(2428 S)
 [11] count:   17,  width:  226 us [191;280]	( 462 S)
 [12] count:    2,  width:  173 us [172;174]	( 354 S)
Level estimates [high, low]:  22954,  12578
RSSI: 1.5 dB SNR: 1.1 dB Noise: -1.1 dB
Frequency offsets [F1, F2]:    3913,  -8146	(+122.3 kHz, -254.6 kHz)
Guessing modulation: No clue...
```

ProfBoc75, 2025-07-10:
@djcarter , just try this (it will be at 433.92 Mhz by default):

    rtl_433 -R 3

From the 2 wav files I got repeating codes of {37}94f00da318, {37}94f00db318, and  {37}94f00db328

When I retried with -y option I got `Prologue-TH` answers, i.e. protocol 3 = ThermoPro TX2.

`rtl_433 -y {37}94f00da318,{37}94f00da318,{37}94f00da318,{37}94f00da318 -M protocol`

[image]

`rtl_433 -y {37}94f00db318,{37}94f00db318,{37}94f00db318,{37}94f00db318 -M protocol`

[image]

`rtl_433 -y {37}94f00db328,{37}94f00db328,{37}94f00db328,{37}94f00db328 -M protocol`

[image]

gdt, 2025-07-16:
I don't understand why you don't just run rtl_433 without specifically setting the decoder list.   Did you try that?

Seems like @ProfBoc75 has pointed out that this decodes, so are you all set as an answered question?   If not please explain.

ProfBoc75, 2025-07-17:
@gdt @djcarter : I identified the data into the wav files. Unfortunately I can't confirm without a cu8 file with metadata (frequency and sample rate). And yes, a simple test is needed with rtl_433 or with -R 3 protocol to confirm that. If it doesn't work, it could be a clone device and with a different modulation, different pulse width.