merbanan/rtl_433 #3333: Help for decoding baldr pool thermometer

endmarsfr, 2025-07-19:
I'm trying to build an rtl-433 decoder for the [baldr pool thermometer](https://www.amazon.fr/dp/B0F7XQZ4NX/?coliid=I2EEO0WNMFGTP0&colid=1YEW6JDWSP18Q&psc=1&ref_=list_c_wl_lv_ov_lig_dp_it) 
I've managed to write the decoder: [thermometre_baldr.txt](https://github.com/user-attachments/files/21328496/thermometre_baldr.txt)

and the decoder is working, but I can't identify the temperature calculation formula.

where i am wrong ? 

captured Octets | Temperature in °C
-- | --
a6 fa 3e 7b ef f7 5f 7f fd | 14.6
a6 fa 3e 7b ef fb 5f 7f ff  | 21.6
a6 fa 3e 7b ef eb bf 7f f8 | 25.5
a6 fa 3e 7b ef ed bf 7f e0 | 26.2
a6 fa 3e 7b ef ff 5f bf d8 | 26.7
a6 fa 3e 7b ef f7 df bf 90 | 26.6
a6 fa 3e 7b ef f5 bf 7f 68 | 27.7
a6 fa 3e 7b ef f5 bf 7f 68 | 27.7
a6 fa 3e 7b ef eb bf 7f 98 | 28.1
a6 fa 3e 7b ef ef 3f 7f bc | 28.5
a6 fa 3e 7b ef eb 7f 7f 8c | 33.4
a6 fa 3e 7b ef ff 3f bf db | 33.8
a6 fa 3e 7b ef e8 fe ff 5c | 34.3
a6 fa 3e 7b ef ee 7f 7f f7 | 34.2
a6 fa 3e 7b ef e8 fe ff ee | 34.4
a6 fa 3e 7b ef eb 7f 7f 9e | 35.2
a6 fa 3e 7b ef ea ff 7f fe | 36.2
a6 fa 3e 7b ef ea ff 7f ec | 37.1
a6 fa 3e 7b ef ee ff bf de | 37.4
a6 fa 3e 7b ef f3 ef 7f f8 | 39.7
a6 fa 3e 7b ef eb ef 7f 40 | 40.5
a6 fa 3e 7b ef f7 f7 bf e8 | 40.8
a6 fa 3e 7b ef fe f7 bf d8 | 40.0
a6 fa 3e 7b ef ff 77 bf d8 | 40.9
a6 fa 3e 7b ef f3 ef 7f e0 | 40.6
a6 fa 3e 7b ef fe f7 bf d8 | 40.0
a6 fa 3e 7b ef fb 3f 7f da | 30.0

zuckschwerdt, 2025-07-19:
You want to use [a BitBench](https://triq.org/bitbench/#c=a6fa3e7beff75f7ffd%5B14.6%5D&c=a6fa3e7beffb5f7fff%5B21.6%5D&c=a6fa3e7befebbf7ff8%5B25.5%5D&c=a6fa3e7befedbf7fe0%5B26.2%5D&c=a6fa3e7befff5fbfd8%5B26.7%5D&c=a6fa3e7beff7dfbf90%5B26.6%5D&c=a6fa3e7beff5bf7f68%5B27.7%5D&c=a6fa3e7befebbf7f98%5B28.1%5D&c=a6fa3e7befef3f7fbc%5B28.5%5D&c=a6fa3e7befeb7f7f8c%5B33.4%5D&c=a6fa3e7befff3fbfdb%5B33.8%5D&c=a6fa3e7befe8feff5c%5B34.3%5D&c=a6fa3e7befee7f7ff7%5B34.2%5D&c=a6fa3e7befe8feffee%5B34.4%5D&c=a6fa3e7befeb7f7f9e%5B35.2%5D&c=a6fa3e7befeaff7ffe%5B36.2%5D&c=a6fa3e7befeaff7fec%5B37.1%5D&c=a6fa3e7befeeffbfde%5B37.4%5D&c=a6fa3e7beff3ef7ff8%5B39.7%5D&c=a6fa3e7befebef7f40%5B40.5%5D&c=a6fa3e7beff7f7bfe8%5B40.8%5D&c=a6fa3e7beffef7bfd8%5B40.0%5D&c=a6fa3e7befff77bfd8%5B40.9%5D&c=a6fa3e7beff3ef7fe0%5B40.6%5D&c=a6fa3e7beffb3f7fda%5B30.0%5D&f=PRE_AND_ID%3F%208h8h8h8h8h%20DATA%208h8h8h%20CHK%3F8h%20&z=1&i=true&cw=4) to format the data. Then look at two temperature values that are close and see what bits change.
E.g. 34.3 to 34.2 is `17 01 00` to `11 80 80` which seems to much change for a single decimal change. Are you sure the coding (PWM) is correct? Do you have a .cu8 sample? (upload here as zip)

endmarsfr, 2025-07-20:
How did you guess that this is a PCM encoding and not PWM? I'm not a signal expert, I trust you, but the analysis of rtl_433 suggested a PWM encoding.

I tried rtl_433 -r file.cu8 -X 'n=baldr,m=OOK_PCM,s=300,l=300,g=700,r=1100,t=150'
and here are the results

```
Temperature | hex code (179 bits)
9.7°C  | 99669655596669559955955695556a6a9555aaaa56658
17.2°C | 996696555966695599559556955599a96555aaaa55a68
19.5°C | 9966965559666955995595569555aa966555aaaa669a8
21.4°C | 9966965559666955995555569555955a6555aaaaa6aa8
21.1°C | 9966965559666955995595569555a6a66555aaaa69aa8
20.1°C | 996696555966695599559556955596666555aaaa596a8
16.3°C | 996696555966695599559556955599696555aaaa55668
19.0°C | 996696555966695599559556955569966555aaaa959a8
19.6°C | 996696555966695599559556955555666555aaaaa69a8
21.0°C | 996696555966695599559556955566a66555aaaa99aa8
21.1°C | 9966965559666955995595569555a6a66555aaaa69aa8
22.4°C | 9966965559666955995595569555a59a6555aaaa6a558
22.9°C | 99669655596669559955955695555a9a6555aaaaa9958
23.1°C | 9966965559666955995595569555aa9a6555aaaa66958
23.7°C | 9966965559666955995595569555666a6555aaaa99658
24.4°C | 9966965559666955995595569555a9aa6555aaaa65a58
25.3°C | 9966965559666955995595569555a955a555aaaaa5598
27.9°C | 996696555966695599559556955569a5a555aaaa65a98
33.2°C | 99669655596669559955955695559996a555aaaa959a8
30.6°C | 99669655596669559955955695556969a555aaaa65668
32.4°C | 9966965559666955995595569555a956a555aaaaa55a8
34.6°C | 99669655596669559955955695556a66a555aaaa666a8
40.0°C | 996696555966695599559556955555955955aaaa9a598
41.5°C | 9966965559666955995595569555a6655955aaaa56698
43.1°C | 9966965559666955995595569555a9595955aaaa59568
44.8°C | 996696555966695599559556955569695955aaaaa5668
44.2°C | 9966965559666955995595569555a6995955aaaa56968
41.6°C | 99669655596669559955955695555a655955aaaa96698
40.4°C | 9966965559666955995595569555a9955955aaaa59998
```

I couldn't figure out how bitbench works, I don't know the decoding format
is there a tutorial that explains step by step what to do?
could you please help me?

[rtl_tests.zip](https://github.com/user-attachments/files/21335365/rtl_tests.zip)

zuckschwerdt, 2025-07-20:
PWM has data only in the pulses, the gaps should all be the same length or a complimentary length to the pulse (i.e. fixed overall length). MC/DMC (sliced as PCM) have data in the gaps, i.e. the gaps will vary and not fit with the pulses.

There are several intricacies to this particular protocol. PCM with MC is correct, but there seems to be one trailing bit that needs to be ignored. Then all bytes a reflected (LSB to MSB). The first byte 'a5' is a sync and ignored in the checksum. The checksum is plain addition, i.e. after reflection `add_bytes(&b[1], 9) == b[10]`. Finally the temperature is given in Fahrenheit with one decimal. See [this BitBench](https://triq.org/bitbench/#c=99669655596669559955955695556a6a9555aaaa56658%5B9.7C%2049.4F%5D&c=996696555966695599559556955599696555aaaa55668%5B16.3C%2061.3F%5D&c=996696555966695599559556955599a96555aaaa55a68%5B17.2C%2062.9F%5D&c=996696555966695599559556955569966555aaaa959a8%5B19.0C%2066.2F%5D&c=9966965559666955995595569555aa966555aaaa669a8%5B19.5C%2067.1F%5D&c=996696555966695599559556955555666555aaaaa69a8%5B19.6C%2067.2F%5D&c=996696555966695599559556955596666555aaaa596a8%5B20.1C%2068.1F%5D&c=996696555966695599559556955566a66555aaaa99aa8%5B21.0C%2069.8F%5D&c=9966965559666955995595569555a6a66555aaaa69aa8%5B21.1C%2069.9F%5D&c=9966965559666955995555569555955a6555aaaaa6aa8%5B21.4C%2070.5F%5D&c=9966965559666955995595569555a59a6555aaaa6a558%5B22.4C%2072.3F%5D&c=99669655596669559955955695555a9a6555aaaaa9958%5B22.9C%2073.2F%5D&c=9966965559666955995595569555aa9a6555aaaa66958%5B23.1C%2073.5F%5D&c=9966965559666955995595569555666a6555aaaa99658%5B23.7C%2074.6F%5D&c=9966965559666955995595569555a9aa6555aaaa65a58%5B24.4C%2075.9F%5D&c=9966965559666955995595569555a955a555aaaaa5598%5B25.3C%2077.5F%5D&c=996696555966695599559556955569a5a555aaaa65a98%5B27.9C%2082.2F%5D&c=99669655596669559955955695556969a555aaaa65668%5B30.6C%2087.0F%5D&c=9966965559666955995595569555a956a555aaaaa55a8%5B32.4C%2090.3F%5D&c=99669655596669559955955695559996a555aaaa959a8%5B33.2C%2091.7F%5D&c=99669655596669559955955695556a66a555aaaa666a8%5B34.6C%2094.2F%5D&c=996696555966695599559556955555955955aaaa9a598%5B40.0C%20104.0F%5D&c=9966965559666955995595569555a9955955aaaa59998%5B40.4C%20104.7F%5D&c=9966965559666955995595569555a6655955aaaa56698%5B41.5C%20106.7F%5D&c=99669655596669559955955695555a655955aaaa96698%5B41.6C%20106.8F%5D&c=9966965559666955995595569555a9595955aaaa59568%5B43.1C%20109.5F%5D&c=9966965559666955995595569555a6995955aaaa56968%5B44.2C%20111.5F%5D&c=996696555966695599559556955569695955aaaaa5668%5B44.8C%20112.6F%5D&f=SYNC%5E8h%20ID%3F%5E8h%5E8h%20%5E8h%5E8h%20%5E8h%5E8h%20TEMP_F%5E12d%20%5E4h%5E8h%20CHKSUM%5E8h%201x&z=1&d=MC&cf=ADD&co=8&cl=72&cw=8).

zuckschwerdt, 2025-07-20:
Next steps: try a variable voltage power supply or known flat batteries to see where the low battery flag is.
If there is a channel selector try to see what changes. Try to remove the batteries for some time to see if the ID changes or is fixed, i.e. "burned-in".
It seems like there is a humidity reading. Try to find out if that is on the display unit or the sensor, i.e. see if you can spot code changes with different humidities.

endmarsfr, 2025-07-21:
Thank you very much for your help
for the battery status, I have tested with weak batteries but, even if they are weak, they are strong enough not to show a low battery status on the devices. I've attached samples where you can see something.
Humidity is displayed on the display unit but not on the sensor (I have temperature on the sensor and the battery indicator too)
I don't have a channel selector on the display unit.

I can see that you were able to decode the temperature using bitbench with manchester coding (GE Thomas)
I tried to do the same decoding in python as bitbench, but I couldn't
I wonder how I could do it in the C decoder for rtl_433?

[rtl_433_tests_battery.zip](https://github.com/user-attachments/files/21343378/rtl_433_tests_battery.zip)

`import binascii

def manchester_decode(bits):
    """Décodage Manchester (GE Thomas) : 01 -> 0, 10 -> 1"""
    decoded = []
    for i in range(0, len(bits), 2):
        pair = bits[i:i + 2]
        if pair == '01':
            decoded.append('0')
        elif pair == '10':
            decoded.append('1')
        else:
            # Gestion des erreurs: si transition invalide, retourne None
            return None
    return ''.join(decoded)

def reverse_bits(bits):
    """Inverse l'ordre des bits dans une chaîne binaire"""
    return bits[::-1]

def decode_temperature(hex_message):
    # Convertir le message hexadécimal en binaire
    binary = bin(int(hex_message, 16))[2:].zfill(len(hex_message) * 4)

    # Décodage Manchester
    decoded_bits = manchester_decode(binary)
    if decoded_bits is None:
        return None, None, "Erreur de décodage Manchester"

    # Prendre les 81 premiers bits (72 bits de données + 8 bits checksum + 1 bit ignoré)
    if len(decoded_bits) < 81:
        return None, None, "Message trop court après décodage"
    full_message = decoded_bits[:81]

    # Séparer les composants
    data_bits = full_message[:72]  # 72 bits de données
    checksum_bits = full_message[72:80]  # 8 bits de checksum
    # Le dernier bit (81e) est ignoré

    # Vérification du checksum (addition avec offset de 8)
    bytes_list = [int(data_bits[i:i + 8], 2) for i in range(0, 72, 8)]
    calculated_checksum = (sum(bytes_list) + 8) % 256
    received_checksum = int(checksum_bits, 2)

    if calculated_checksum != received_checksum:
        return None, None, f"Checksum invalide ({calculated_checksum} vs {received_checksum})"

    # Extraction des 12 bits de température (bits 60 à 71 du message complet)
    temp_bits = data_bits[60:72]
    temp_bits_reversed = reverse_bits(temp_bits)  # Inversion des bits
    temp_f = int(temp_bits_reversed, 2) / 10.0  # Conversion en float

    # Conversion Fahrenheit -> Celsius
    temp_c = (temp_f - 32) * 5 / 9

    return temp_f, round(temp_c, 1), "Succès"

# Exemple avec votre message
hex_message = "99669655596669559955955695556a6a9555aaaa56658"
temp_f, temp_c, status = decode_temperature(hex_message)

if temp_f is not None:
    print(f"Température: {temp_f} °F, {temp_c} °C")
    print(f"Statut: {status}")
else:
    print(f"Erreur: {status}")`

[rtl_433_tests_battery.zip](https://github.com/user-attachments/files/21343379/rtl_433_tests_battery.zip)

endmarsfr, 2025-07-21:
here's my implementation of baldr thermometer temperature decoding
I don't know if this is the right way to integrate it into rtl_433, but this version decodes temperature correctly.
If it's right, we can continue to decode battery information from this base.

[thermometre_baldr.c.txt](https://github.com/user-attachments/files/21351274/thermometre_baldr.c.txt)

zuckschwerdt, 2025-07-21:
There is already a reflect function: https://github.com/merbanan/rtl_433/blob/master/include/bit_util.h#L28
You must not convert the units but output Fahrenheit and then use `-C si`

You should not need string.h

Don't use memset() but e.g. `uint8_t decoded_bytes[EXPECTED_BYTES] = {0};`

There is already a MC decode function: https://github.com/merbanan/rtl_433/blob/master/include/bitbuffer.h#L122

ProfBoc75, 2025-07-24:
> I tried the MC decoding function of RTL_433, but unfortunatelly, no temp displayed

I will review your decoder and help you, in the meantime, check all other decoders into /src/devices folder, this can help you to build your own decoder following rtl_433 standard.

ProfBoc75, 2025-07-24:
@endmarsfr 

My findings:
- Your sensor reference is Baldr HCS528ARF Pool Thermometer.

The protocol is very similar to rainpoint_hcs012arf.c (Rain gauge sensor)

Steps to get the figures:

- Check if repeated rows
- Use bitbuffer_manchester_decode function to get the Manchester IEEE 802.3 decoding
- Use bitbuffer_invert to get the Manchester G.E. Thomas decoding
- Use reflect_bytes function to reflect all the bytes of the message.
- Check if the first byte is 0xa5
- Calculate the sum of bytes and check it, i.e. Checksum.
- Extract the ID, because of the reflect, my assumption is that the data are also little-endian coded
- Extract the Temp Raw value, for sure little endian coded
- Extract battery information, my assumption is the bat flag position is the same as for rainpoin_hcs012arf

And tada;

<details>

<summary>baldr_hcs528arf.c</summary>

```c
/** @file
    Baldr HCS528ARF Pool Thermometer sensor.

    Copyright (C) 2025 Bruno OCTAU, Christian W. Zuckschwerdt, \@endmarsfr

    This program is free software; you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation; either version 2 of the License, or
    (at your option) any later version.
*/
/**
Baldr HCS528ARF Pool Thermometer sensor.

Brand:

- Fujian Baldr Technology Co., Ltd.

Reference:

- Baldr HCS528ARF Pool Thermometer sensor works with Baldr HCS015T2H Display Station

S.a. Issue 3333

OOK PCM, MC, Invert and Reflect, the message is repeated 10 times
The protocol is very similar to Rainpoint HCS012ARF, here 11 bytes instead of 10.

Flex:

    rtl_433 -X 'n=name,m=OOK_PCM,s=300,l=300,g=700,r=1100'

Samples:

    data      : 9966965559666955995595569555a6a66555aaaa69aa8
    MC decoded  a5 90 25 60 a0 81 80 dd 40 ff 6f 1
    MC Zerobit  25 90 25 60 a0 81 80 95 40 ff 27 8

Data Layout (before reflect):

    Byte Position  0  1  2  3  4  5  6  7  8  9 10
    Sample        a5 90 25 60 a0 81 80 dd 40 ff 6f 1
    Data          SS II II II II FB FF TT TF FF CC T

- SS:{8}  0xa5 , header sync word (OOK PCM then MC decoded) or 0x25 if directly OOK MC Zerobit decoded
- II:{48} Sensor ID ?
- FF:{x}  Fixed value
- B1:{1}  Low Battery flag = 1, Good Battery = 0  (To be confirmed, assumption from rainpoint hcs012arf)
- B2:{1}  Powered on, batteries are inserted = 1, then always = 0  (To be confirmed, assumption from rainpoint hcs012arf)
- TT:{12} Temperature, Fahrenheit, scale 10
- CC:{8}  Checksum, addition of previous reflected bytes except sync word.

*/

#include "decoder.h"

static int baldr_hcs528arf_decode(r_device *decoder, bitbuffer_t *bitbuffer)
{

    bitbuffer_t decoded = { 0 };

    // Find repeats
    int row = bitbuffer_find_repeated_row(bitbuffer, 4, 179);
    if (row < 0)
        return DECODE_ABORT_EARLY;

    if (bitbuffer->bits_per_row[row] > 179)
        return DECODE_ABORT_LENGTH;

    bitbuffer_manchester_decode(bitbuffer, row, 0, &decoded, 11 * 2 * 8);
    bitbuffer_invert(&decoded);

    uint8_t *b = decoded.bb[0];
    reflect_bytes(b, 11);

    decoder_log_bitrow(decoder, 2, __func__, b, 88, "MSG");

    if (b[0] != 0xa5) {// Header sync word
        decoder_logf(decoder, 2, __func__, "Wrong Sync Word %02x expected 0xa5", b[0]);
        return DECODE_ABORT_EARLY;
    }

    // Checksum
    int sum = add_bytes(&b[1], 9); // header not part of the sum
    if ((sum & 0xff) != b[10]) {
        decoder_logf(decoder, 2, __func__, "Checksum failed %04x vs %04x", b[10], sum);
        return DECODE_FAIL_MIC;
    }

    int id       = b[4] << 24 | b[3] << 16 | b[2] << 8 | b[1]; // little endian
    int bat_low  = (b[5] & 0x02) >> 1;
    int temp_raw = (b[8] & 0x0F) << 8 | b[7]; // little endian
    float temp_F = temp_raw * 0.1f;

    /* clang-format off */
    data_t *data = data_make(
            "model",            "",             DATA_STRING,    "Baldr-HCS528ARF",
            "id",               "",             DATA_FORMAT,    "%08x",   DATA_INT,       id,
            "battery_ok",       "Battery",      DATA_INT,       !bat_low,
            "temperature_F",    "Temperature",  DATA_FORMAT,    "%.1f F", DATA_DOUBLE, temp_F,
            "mic",              "Integrity",    DATA_STRING, "CHECKSUM",
            NULL);
    /* clang-format on */

    decoder_output_data(decoder, data);
    return 1;
}

static char const *const output_fields[] = {
        "model",
        "id",
        "battery_ok",
        "temperature_F",
        NULL,
};

r_device const baldr_hcs528arf = {
        .name        = "Baldr HCS528ARF Pool Thermometer sensor",
        .modulation  = OOK_PULSE_PCM,
        .short_width = 320,
        .long_width  = 320,
        .gap_limit   = 700,
        .reset_limit = 1000,
        .decode_fn   = &baldr_hcs528arf_decode,
        .fields      = output_fields,
};
```

I tried with your samples and it works fine, few samples are not good and not decoded, even with flex decoder.

If you want I can create a PR and Add support for Baldr HCS528ARF Pool Thermometer. Let me know.

ProfBoc75, 2025-07-26:
> Thank you very much, it works for me too. Can this new device be added to the list of devices supported by rtl_433?
> 
> for the parameters should we use these ones
> 
> ```
> .modulation  = OOK_PULSE_PCM,
> .short_width = 320,
> .long_width = 320,
> .gap_limit = 700,
> .reset_limit = 1000,
> ```
> 
> or these ones?
> 
> ```
>     .modulation  = OOK_PULSE_PCM,
>     .short_width = 300,
>     .long_width  = 300,
>     .gap_limit   = 700,
>     .reset_limit = 1100,
>     .tolerance   = 150,
> ```

Bonjour @endmarsfr , yes I can add the support of it.

About the pulse width, it's a about success, I did some check from your cu8 files and compared with the other rain gauge sensor decoder, and looks like the pulse width is more around 320 us than 300 us, but the OOK PCM decoder is robust and it works in both case. 

The thermometer sensor should send information every minute or 3 minutes or so. So, you need to test the decoder with a first width around 320 us for one hour or one day and check if all the messages are decoded by rtl_433. Then test another hour / day with the other values, and compare the success numbers.

Notice that it's normal that few messages are not decoded, it depends if you have lot of "rf noise" around you, I mean, if there are lot of sensors around you and neighborhood, then you may have conflict and the signal can't be decoded.

endmarsfr, 2025-07-27:
I tested baldr_hcs528arf.c, and it worked fine.
You can now add it to the RTL_433 supported list.
Regards,

endmarsfr, 2025-11-09:
Hi,
When  baldr_hcs528arf.c will be added to device list ?
regards

drwuke, 2026-11-15:
**BaldrTherm HCS706 pool thermometer — protocol decoded**

Building on the HCS528ARF work in this issue, the HCS706 pool sensor
(433.92 MHz OOK, solar panel + LCD, 3×AAA) uses the same family encoding
with a longer frame. I captured bursts with a CC1101 receiver and verified
the decode against the vendor (HomGar) cloud readings during a controlled
ice-water cooldown — 7 out of 7 packets matched exactly.

Frame format:

- PCM ~305 µs, Manchester (GE Thomas: 10→1, 01→0), one frame repeated
  10× per burst
- Manchester decode gives 114 bits; skip the first 2 bits → 14 bytes;
  reflect (bit-reverse) each byte
- Byte 0: sync `0x55`
- Bytes 1–4: device ID, little-endian
- Bytes 5–10: flags/counters (not fully decoded yet; byte 5 ∈ {0x81, 0x88,
  0x89}, byte 8 = 0x4f, byte 9 = 0x01, byte 10 = 0x0b in my captures)
- Bytes 11–12: temperature, 12-bit little-endian
  (byte 11 = low 8 bits, low nibble of byte 12 = high 4 bits);
  value = temperature in °F × 10
- Byte 13: checksum = sum of bytes 1–12, mod 256

Worked example (after Manchester decode + reflect):
`55 0c c8 00 56 81 00 10 4f 01 0b e9 02 XX`
→ ID 0x5600C80C, temperature raw 0x2e9 = 745 → 74.5 °F = 23.6 °C,
checksum = sum(bytes 1–12) mod 256.

A working decoder mirroring `baldr_hcs528arf` has been added to
rtl_433_ESP and verified live in OpenMQTTGateway: Home Assistant now
receives temperature / RSSI / battery from a local gateway, no cloud
involved. Happy to open a PR if useful.