merbanan/rtl_433 #3334: Reolink Chime v2

NHellFire, 2025-07-25:
I've been looking at the new [Reolink chime](https://reolink.com/product/reolink-chime/). This one is 868MHz and seems to be a different encoding.

Any help with a decoder would be appreciated.

Attached captures, made with `rtl_433-rtlsdr.exe -f 868M -s 1024k -S unknown`

Thanks!

klohner, 2025-07-27:
FWIW, I can get data from these samples with a flex decoder like this:

`rtl_433 -X "n=Chime_v2,m=FSK_PCM,s=40,l=40,r=500,preamble={16}5593" -Y minmax -r .\g022_868M_1024k.cu8`

Here's a [BitBench](https://triq.org/bitbench/#c=%7B114%7D0b51def7e91d9aed693aa7d9e3d78%20%5Bg017_868M_1024k.cu8%5D&c=%7B306%7D0b51dedfc11c9aed693aa7d97bd23970c68aba67eadac957118d7ba83837432db1dcc66e0abb4%20%5Bg018_868M_1024k.cu8%5D&c=%7B210%7D0b51deebf51f9aed693aa7d92b0f8943673c53b22cbd4da161294%20%5Bg018_868M_1024k.cu8%5D&c=%7B626%7D0b51deb7a91e3df0693aa7d92b0f894395570a549c4650a96c1a6b9919b91477b5f649afc84f0f0996b3dcfac59d9c92bd0be3c44961569afab975b82115176683ad9a1d4c9492a4ed4b1ed73de38%20%5Bg020_868M_1024k.cu8%5D&c=%7B242%7D0b51dee7f91e3df0693aa7d92b0f894394570a549a4652a96c1a5b9931e84%20%5Bg021_868M_1024k.cu8%5D&c=%7B626%7D0b51deb7a91e3df0693aa7d92b0f894393570a549c4654a96c1a6b9919b91477b5f649afc84f0f0996b3dcfac59d9c92bd0be3c44961569afab975b82115176683ad9a1d4c9492a4ed4b1ed78ecb8%20%5Bg022_868M_1024k.cu8%5D&f=h&z=1&cw=4) of the data.

zuckschwerdt, 2025-07-27:
That looks like somewhat stable data, but way to much bits for a doorbell, strange.

mgg1010, 2025-12-31:
Hi - I'd be interested in getting this working too.  I don't understand "flex def" - what is needed to progress this?