merbanan/rtl_433 #3342: TMPS sensor support -  Sefis M3, Careud, Sykik SRTP300

ivantichy, 2025-08-03:
Hello.

spent a few days trying to solve this beast: 

[image]

Similar can be found as Sefis M3, Careud, Sykik SRTP300

Internally it uses SP400 with 8051 procesor. [datasheet](https://www.infineon.com/assets/row/public/documents/24/49/infineon-sp400-11-01-datasheet-en.pdf)

Seems like FKS_PCM with 52us. So far I was not able to find the patters. However I kept it on a table and recorded messages - two are very close to each other (might be small temp change).

Still playing with it however any help is really appreciated.

[Bitbench](https://triq.net/bitbench#c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2096%205A%20A5%2056%2065%2095%20AA%2056%20A9%2096%2056%2065%2059%20A5%2066%2056%20A6%2066%20E%20%5Ba%20rear%200bar%2026temp%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2096%2096%20A5%206A%2065%2095%20AA%206A%20A9%20AA%2056%2059%2059%2055%20A6%2056%2069%2096%20E%20%5Bc%20rear%200bar%2024temp%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%20AA%20AA%2055%20AA%20A6%2056%2059%20A6%2066%2066%2099%2099%2095%20A9%20A5%2065%20A5%2065%20E%20%5Bb%20rear%202.6bar%2025temp%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2055%20A9%2055%2055%20A6%2056%2059%20A5%2066%2065%2099%2096%2095%20A9%2095%2059%20A9%2096%20E%20%5Bd%20rear%202.6bar%2024temp%5D&c=&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%206A%2069%209A%20A5%2066%206A%20A9%2066%20A9%20A5%2055%2065%209A%2059%2055%2056%2065%20AA%20E%20%5Bf%20front%200bar%2025temp%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%20A9%209A%2059%205A%20A9%20AA%2065%2099%2066%2056%2095%20A6%209A%2069%205A%205A%20A5%2056%20E%20%5Be%20front%202.6bar%2026temp%5D%20&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%20A9%20A9%2059%2065%20A9%20AA%2065%20A6%2066%2065%2095%20A5%209A%2059%2065%209A%2065%20AA%20E%20%5Bg%20front%202.6bar%2025temp%5D&c=&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2096%2096%20A5%206A%2065%2095%20A9%206A%20A9%20AA%2055%2059%2056%2055%20AA%20AA%20A9%20A9%20F%20%5Bx%20rear%200bar%20%3Ftemp%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2096%2095%20A5%2069%2065%2095%20A9%2069%20A9%20A9%2055%205A%2056%2059%2066%2055%20A6%2095%20F%20%5Bx%20rear%200bar%20%3Ftemp%5D&c=&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2096%2066%20A5%205A%2065%2095%20A9%205A%20A9%209A%2055%2069%2056%2095%20AA%2055%20A6%2096%20F%20%5Bx%20rear%20no%20movement%20for%20a%20long%20time%2C%20temp%20might%20change%200bar%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2095%2065%20A6%2059%2066%2095%20AA%2059%20AA%2099%2055%206A%205A%2099%2055%2099%20A9%20A5%20F%20%5Bx%20rear%20heated%20up%200bar%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2095%20A9%20A6%20A5%2065%2095%2055%20A5%20AA%2065%2055%2096%2059%2099%2069%2099%206A%205A%20F%20%5Bx%20rear%20cooling%20down%2C%20movement%2C%200bar%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2096%2065%20A5%2059%2065%2095%20A9%2059%20A9%2099%2055%206A%2059%2099%2069%2096%209A%20A5%20F%20%5Bx%20rear%20no%20movement%2C%200bar%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2096%2066%20A5%205A%2065%2095%20A9%205A%20A9%209A%2055%2069%2056%2095%20AA%2055%20A6%2096%20F%20%5Bx%20rear%20no%20movement%2C%200bar%5D&c=AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%20AA%2066%2099%2096%20A6%2096%2066%20A5%205A%2065%2095%20A9%205A%20A9%209A%2055%2069%2056%2095%20AA%2055%20A6%2096%20F%20%5Bx%20rear%20no%20movement%2C%200bar%5D&f=d%0A%0A&z=1&a=Preamble&m=&d=MC&cw=4)

[triq](https://triq.org/pdv/#AAB10400340068011000008080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080819190819191809080819081809081808090818080908180918091808090808180908190808190808180808080808090818080918091809191808080908190818080A355)

ivantichy, 2025-08-03:
Link to cu files: [here](https://drive.google.com/drive/folders/1tQ2RRxlq9M1HJc05n5wPOhksOdlyP5ra?usp=sharing)

Names correspond to "a", "b", etc. in bitbench comments.

ivantichy, 2025-08-03:
Yes, from and rear. 

The “missing” id is weird. Sensors have some number printed on them. I can post it tomorrow. 

The order in first part is a,b,c,d…

The part with “x” is in order but some messages might be missing. 

Nice tip with the pressure.

ProfBoc75, 2025-08-10:
Hi @ivantichy , I have a finding, the CRC is classic CRC-16/XMODEM, poly=0x1021  init=0x0000, without the preamble 0xff and the sync word 0x5a9d, only on the Data values.

Data Layout:

            Preamble ___________________________________ SyncWord  Data _______________   CRC 16
    Sample [ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff] [5a 9d] [93 c1 48 f1 e9 14 2c] [51 d5]

But looks like the temp, pressure, id and other flags are mixed into the Data field, not possible to guess for the moment.

ivantichy, 2025-08-13:
Thank you @ProfBoc75 for the CRC info. That helped.

As far as I am on vacation with no computer I just played with ChatGPT. 

It seems pressure is in b4 and b5. The internal ADC is 13bit only so not sure whether/how they use the last 3LBS which have different values. 

[image]

ivantichy, 2025-10-19:
So, I implemented transmitter via CC1101 including CRC calculation and I was able to send and successfully read (on original display) the very same values.
Problem came when I tried to change any of bits and recalculated CCR- usually message was ignored, or "H.H" as a pressure was shown or some weird values. List:

Format: data pressure temp rear/front (not encoded)
// 99 c7 48 e7 ef 12 20 0.4 40c r
// 99 c7 48 f7 ef 12 20 0.0 24c r
// 76 bc 57 e5 ec 04 b2 0.0 25c f
// 99 c7 48 f7 ef 12 01 HH 24c r
// 99 c7 48 f7 ef 12 11 HH 24c r
// 99 c7 48 f7 ef 12 12 HH 24c r
// 99 c7 48 f7 ef 12 13 6.4 24c r
// 99 c7 48 f7 ef 12 14 0.0 24c r
// 99 c7 48 f7 ef 12 15 HH 24c r
// 99 c7 48 f7 ef 12 16 HH 24c r
// 99 c7 48 f7 ef 12 17 6.4 24c r
// 99 c7 48 f7 ef 12 18 0.0 24c r
// 99 c7 48 f7 ef 12 19 HH 24c r

Then I tried to use the "pressure spraye" with there readings (rear sensor, encoded)
ffffffffffffffffffffffffffffff5a9df05ed07e567be2edc3
[2.6 26]
ffffffffffffffffffffffffffffff5a9dff51e28d4568f090b5
[2.3 24]
ffffffffffffffffffffffffffffff5a9dfe50e18c4469ee02ec
[2.2 24]
ffffffffffffffffffffffffffffff5a9de799fab5bd502025fe
[2.1 24]
ffffffffffffffffffffffffffffff5a9de698f9b4bc512ea596
[2.0 24]
ffffffffffffffffffffffffffffff5a9de5a3f4c3bb5618e98a
[1.9 24]
ffffffffffffffffffffffffffffff5a9deea0f1dcb4596ea205
[1.8 24]
ffffffffffffffffffffffffffffff5a9dda9cfdc8b04d5e6178
[1.7 24]
ffffffffffffffffffffffffffffff5a9dd3f50611a94438be99
[1.6 24]
ffffffffffffffffffffffffffffff5a9dd2f40510a8453ebff9
[1.5 24]
ffffffffffffffffffffffffffffff5a9dcb8d0eb9a13c587bfe
[1.4 24]
ffffffffffffffffffffffffffffff5a9dcc8a0bbaa23f5a3e31
[1.3 24]
ffffffffffffffffffffffffffffff5a9dc88617a69e33525b51
[1.2 24]
ffffffffffffffffffffffffffffff5a9dc58314a39b36581582
[1.1 24]
ffffffffffffffffffffffffffffff5a9db19f10bf973a7009b7
[1.0 24]
ffffffffffffffffffffffffffffff5a9dba7c1da8902d9e6f24
[0.9 24]
ffffffffffffffffffffffffffffff5a9db66829848c218ebda0
[0.8 24]
ffffffffffffffffffffffffffffff5a9db59324b38b2678f85c
[0.7 24]
ffffffffffffffffffffffffffffff5a9da1af20cf872a503ba9
[0.6 24]
ffffffffffffffffffffffffffffff5a9daa6c2d98801d9e242a
[0.5 24]
ffffffffffffffffffffffffffffff5a9da9d738f7ff1220cc85
[0.4 24]
ffffffffffffffffffffffffffffff5a9da4e23302fa171a06e9
[0.3 24]
ffffffffffffffffffffffffffffff5a9d90ee3f0ef60be20c30
[0.2 24]
ffffffffffffffffffffffffffffff5a9d9cda3b0af20fda93de
[0.1 24]