merbanan/rtl_433 #3343: Funny line format in SYSLOG

alanrothenbush, 2025-08-09:
Attempting to use the example script

  `rtl_433_custom.py`

on an RPi, after installing RSYSLOG.  Started rtl_433, using the command in the script

  `rtl_433 -F syslog::1433`

As with most scripts, a line is "discovered" using the code

  `if line.startswith("<"):`

However, the lines from rtl_433 appearing in SYSLOG do not begin with the "<" character.  They start with the time.  This is troublesome.

zuckschwerdt, 2025-08-11:
A syslog program will usually strip the header and just print the text. Our example scripts are not meant to work with processed syslog output. The raw syslog directly form rtl_433 should be used.

zuckschwerdt, 2025-08-11:
The rtl_433 UDP output is RFC 5424 (IETF-syslog protocol).
Test it with e.g. `socat udp-listen:1433 -`

zuckschwerdt, 2025-08-11:
I think you don't want to use a syslog daemon. rtl_433 just uses the syslog protocol, but usually not the daemon. If something like `tail -f syslog` works then your messges won't reach the custom script. Use `-F syslog:127.0.0.1:1433` and run the script.

closed 2025-10-20 by gdt