merbanan/rtl_433 #3345: Fan Remote with unusual start of frame pulse+pause?

thehans, 2025-08-10:
I have a remote control for a ceiling fan with light.  The brand of fan is Sofucor, and the remote is Model FT0317R from Shenzen Funpower General Technology Co Ltd.

I have recorded all the button presses in urh and am able to decode them there, but I can't get consistent decoding from rtl_433.  I think this is at least partly due to the start of each "message frame"  (not sure if that is 100% accurate terminology) being atypical and throwing off decoding.  
It is basically using `OOK_PWM` but each frame starts with a single short pulse and then 0% duty for 3 periods before beginning the actual message.  

```
PWM is on a period of 1212us
'0' = Short 25% duty  (302us ON time)
'1' = Long 75% duty  (906us ON time)
'-' =  0% duty (1212us of nothing)

Start of frame = "0---"
Then 16 bits of button id (its really just 8 bits repeated)
Then 16 bits which never change across messages, I guess this is a hardcoded unique remote ID.
This whole frame is repeated 4 more times (total of 5) for each button press.
```
I can't get rtl_433 to make any sense of these signals even though I think my understanding of their structure is pretty complete.

The webviewer also seems to mark short pulses as 1's and long pulses as logic 0's ... Or based on its choice of placement it seems to be counting the short/long durations of the *off* time, which is perfectly inverse: a short signal is off for the same time a long signal is on.  Is this normal?
https://triq.org/pdv/#AAB01006010000012E038B005411B902CF839455+AAB02F06040000012E038B005411B902CF929292A1929292A1929292A1929292A1929292A19292A1A192A192A192A1A1A19455+AAB02E06010000012E038B005411B902CF929292A1929292A1929292A1929292A1929292A19292A1A192A192A192A1A1A555

The full detail for every button press is below
```
Each frame = 36bits(or periods) * 1212us =    
Hex message format: S BBBB MMMM (always repeated 5 times in entirety with no delay between)
  S = 4 period start, short pulse then pause = 0---
  BBBB = 8bit button code, doubled
  MMMM = 16 bit unchanging hardcoded remote ID? (no dip switches)
Because of repeated frames with no delay, the short pulse start of the next frame with a pause after makes it appear to be grouped with the previous frame, but it is not.  Otherwise the last of repeated frames would contain a final bit like this and it does not.  

All 16 buttons:
S 1111 1357 x5 Fan On/Off 
S 0c0c 1357 x5 1H Fan cutoff 
S 0a0a 1357 x5 4H Fan cutoff
S 0d0d 1357 x5 8H Fan auto-off
S 0e0e 1357 x5 Temp-based fan speed
S 0f0f 1357 x5 Breeze mode (cycles fan speed over time)
S 0202 1357 x5 Fan Speed 1
S 0303 1357 x5 Fan Speed 2
S 0404 1357 x5 Fan Speed 3
S 0505 1357 x5 Fan Speed 4
S 0606 1357 x5 Fan Speed 5
S 0707 1357 x5 Fan Speed 6
S 0909 1357 x5 Fan Reverse
S 0000 1357 x5 Light Brightness +
S 0808 1357 x5 Light On/Off
S 0101 1357 x5 Light Brightness -
```

A highlighted single frame, the 1st of 5, demonstrating the bit boundaries. 
[image]

ProfBoc75, 2025-08-12:
@thehans Try with the gap at 4500 us like that:

    rtl_433 -X "n=FanRemote,m=OOK_PWM,s=301,l=906,r=1500,g=4500"

You should have the repeated frames. Try different values for the reset, I'm not sure about the 1500 us.
Share here a cu8 sample capture with `-S unknown` to get the exact modulation.

zuckschwerdt, 2025-08-12:
There is no way to distinguish the preamble bit from a last packet bit if it's attached with no gap.
We are used to this with the EV1527 senders, data is 24 bits but will appear as 25 bits with the next wake-up bit attached to the end of data.
Data rows here really should look like you have above, 1 / 33 / 33 / 33 / 33 / 32. Gap should be 1000, reset 5000, i.e. `rtl_433 -X "n=FanRemote,m=OOK_PWM,s=300,l=900,g=1000,r=5000"`

thehans, 2025-08-12:
@zuckschwerdt Even if I accept that preambles are just impossible to interpret reasonably, is it typical to count short pulses as 1s and long pulses as 0s?  Because its still doing that.

```
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR with TLS
[Input] Test mode active. Reading samples from file: /home/hans/projects/sdr/FanRemote433/g001_434.07M_250k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.231060s
model     : FanRemote    count     : 6             num_rows  : 6             rows      : 
len       : 1            data      : 8, 
len       : 33           data      : eeeeeca88, 
len       : 33           data      : eeeeeca88, 
len       : 33           data      : eeeeeca88, 
len       : 33           data      : eeeeeca88, 
len       : 32           data      : eeeeeca8
codes     : {1}8, {33}eeeeeca88, {33}eeeeeca88, {33}eeeeeca88, {33}eeeeeca88, {32}eeeeeca8
```

ckuethe, 2026-10-25:
I think it's Q&A: "I have this remote and I'm not sure how to get consistent decodes." Which is relevant to my interests. I was just about to start doing this exact thing for my Sofucor fan.

Also I note that someone else is working on this. According to https://github.com/esphome/esphome/commit/7794946039f7be685c76713d8fd7edf75ca8ac8a you might be able to decode with
```
// rtl_433 -r sofucorfan.ook -X 'n=SofucorFan,m=OOK_PPM,s=564,l=1684,reset=4516'
// 32 bit data returned.  First 2 bytes appear to be address
// Second 2 bytes appear to be a command - the last byte being the inverse of the previous.
```

There are different timings in https://github.com/tclancy/ook-signal-explorer/pull/1/changes#diff-aeb9ac839e0f05047b78fe65fa38baef593185ecf8cbad661983c4d39158f751

```
frequency_mhz: 315.4
encoding: OOK_PDM  # Pulse Distance Modulation

timing:
  sync_us: 8000       # sync HIGH pulse at start of each repetition
  sync_gap_us: 670    # LOW gap between sync and first data bit (tune if fan doesn't respond)
  pulse_us: 400       # carrier-ON duration, same for all bits
  zero_gap_us: 670    # LOW gap after pulse = bit 0
  one_gap_us: 1800    # LOW gap after pulse = bit 1
  repeat_count: 20    # firmware sends 20×; real remote sends 36–41×
```

I don't have my rtlsdr handy so I can't test that at the moment.

merbanan, 2026-11-08:
Closing as this is a Q&A rather than a bug or a concrete device support request. As noted above, the ESPHome project has work on this Sofucor/Funpower FT0317R remote that may help. If you want rtl_433 support, please open a new issue with a few `.cu8` captures (`-S unknown`) and the button labels, or a PR with a flex decoder conf.

closed 2026-11-08 by merbanan