merbanan/rtl_433 #3357: Cannot get rtl_433 (next) to connect to Mosquito broker

DP042715, 2025-09-09:
As described [here](https://community.home-assistant.io/t/cant-get-rtl-433-next-to-connect-auth-to-mqtt/929120) in this home assistant community thread, while modernity will allow connections from MQTT Explorer or the auto discovery add on, it’s simply refusing to connect to the IP address associated with rtl_433. 

Any suggestions?

DP042715, 2025-09-09:
@zuckschwerdt in my last attempt to figure this out with ChatGPT it recommended to set up the config for rtl_433 as follows:
```
rtl_433_conf_file: ""
retain: true
device: 0
convert: si
protocol: 40
frequency: 433.92M
log_level: debug
output_format: json
```
and instead it would use the Auto Discovery add-on as the bridge to connect to MQTT:
```
master: false
mqtt_host: core-mosquitto
mqtt_port: 1883
mqtt_user: mqtt_user
mqtt_password: mqtt_password
mqtt_retain: true
rtl_topic: rtl_433
device_topic_suffix: ""
discovery_prefix: homeassistant
discovery_interval: 600
force_update: true
log_level: debug
```

As you can see the password is quite simple/standard.  Maybe simplifying it more to not include the underscore `_` might help? But I'm open to trying any changes to either configs here.

gdt, 2025-09-09:
I personally, am not ok with debugging code written by LLMs, especially when those LLMs are trained on open-source code in violation of the licenses.   I suspect we haven't really talked about norms surrounding LLMs -- so this is just me expressing my personal take.

zuckschwerdt, 2025-09-10:
The underscore is not a problem. That's not a rtl_433 config though. See https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf
You want something like `output_format: mqtt://172.30.32.1:1883,user=mqtt_user,pass=mqtt_password`
Best to ask the add-on project for support with that config.